edk2-porting/WOA-Drivers #2: Driver cannot be fully generated.

longjunyu2, 2021-07-22:
It seems that the driver package of the last line in definitions will not be put into the output folder.
``` shell
[start]
\components\SDM845\Platform\minimal
\components\SDM845\Platform\GPIO
......
\components\SDM845\Platform\FG
[end]
```
And FG won't be contained in the output.
A simple way to solve this problem is to add a new line in the file of definitions to the end.
``` shell
[start]
\components\SDM845\Platform\minimal
\components\SDM845\Platform\GPIO
......
\components\SDM845\Platform\FG
  
[end]
```
But the essence of the problem may be a problem of circular reading.

And thank you for providing this amazing tool which is very useful. It effectively classifies the available drivers, so I hope it can become better.

indianets, 2021-07-22:
Which platform you got this issue on?

BigfootACA, 2021-07-23:
I found this problem when I was writing the script, and I think we should make sure that there is a closing line in the file. But it seems that the closing line is often missed

BigfootACA, 2021-09-21:
already fixed

closed 2021-09-21 by BigfootACA