edk2-porting/WOA-Drivers #8: Beryllium: Cannot boot into Win10

RusJJ, 2021-08-18:
Hi there! Installed the latest **MINIMUM** driver. Experiencing crash "IRQL LESS OR EQUAL".
**WITHOUT** drivers is bootable but OTG IS **NOT WORKING**.

RusJJ, 2021-08-18:
Btw with all shipped drivers OS is completely freezes at loading screen (with 5 white circles) and Phone is rebooting

RusJJ, 2021-08-19:
Okay, it looks like that Windows version is kinda outdated. Got the latest ISO image: now im able to boot. But there's a problem: i cannot boot into a setup window, infinitely loading

Idonotkno, 2021-08-19:
@RusJJ Please take a look at the tutorial. The use of Windows Installer on our device may cause the device to become brick, so it needs to be installed manually by dism/dism++

RusJJ, 2021-08-19:
> @RusJJ Please take a look at the tutorial. The use of Windows Installer on our device may cause the device to become brick, so it needs to be installed manually by dism/dism++

Steps: Use DISM++ to install Windows, Use DISM++ to install drivers, Do some things to disable driver signature verification, BOOT IN
Result: Infinite loading **OF WINDOWS**

Idonotkno, 2021-08-19:
@RusJJ Wrong boot or uefi image

RusJJ, 2021-08-19:
> @RusJJ Wrong boot or uefi image

Wrong Boot image? wat
Wrong UEFI won't let me even load into the OS.

Idonotkno, 2021-08-19:
@RusJJ efi or uefi boot

RusJJ, 2021-08-23:
Old Windows. Fixed with the latest Windows 11 (for example)

closed 2021-08-23 by RusJJ