edk2-porting/WOA-Drivers #10: beryllium drivers not installing

salam-siddique, 2021-08-22:
I dont know why but the drivers are not installing when in PE environment

strongtz, 2021-08-22:
@salam-siddique probably you didn't execute extract.sh

BigfootACA, 2021-08-23:
you need run extract.sh or extract.ps1

salam-siddique, 2021-08-23:
i am not a developer, just an enthusiast so i might seem a bit lost.
But when do i have to run it?
In the PE environment?

Thanks in advance.

BigfootACA, 2021-08-23:
run on the host, not on pe

TAOcroatia, 2021-08-23:
On your computer, open the WOA-Drivers folder, ctrl+right click, open in powershell/terminal, execute ```
. /extract.ps1 beryllium
```
New folder `output` will be created with the drivers

salam-siddique, 2021-08-23:
did as suggested but shows as below:

#!/bin/bash
set -e
if [ -z "${1}" ]
then
	echo "no codename specified" >&2
	echo "Usage: ${0} <CODENAME>" >&2
	exit 1
fi
cd "$(dirname "$0")"
CONFIGS=definitions
CONFIG="${CONFIGS}/${1}.txt"
if ! [ -f "${CONFIG}" ]
then
	echo "ERROR: your model has no definition file, please check!" >&2
	exit 1
fi
rm -rf ./output
echo "copying drivers..."
while read -r line
do
	file="${line//$'\r'/}"
	file="${file//\\/\/}"
	cp -r ."${file}" output/
done<"${CONFIG}"
echo "rename drivers..."
find output -type f -name '*.inf_'|while read -r line
do mv "${line}" "${line//.inf_/.inf}"
done
echo "done"

And no output folder created

salam-siddique, 2021-08-23:
i ran the below script

./extract.sh beryllium

salam-siddique, 2021-08-23:
did it succesfully now

thanks for the help, actually the system was not running unsigned scripts

i got it changed so that it could run unsigned scripts.

Thanks for help everyone

salam-siddique, 2021-08-23:
one more thing, now that i have the drivers, how do i boot in the PE environment again?
Or do i have to do the whole process by formatting the drives and reloading the windows image again?

Thanks in advance

BigfootACA, 2021-08-23:
rename EFI folder of esp partition

closed 2021-08-23 by BigfootACA

salam-siddique, 2021-08-23:
And how do i do it?

cvamoose, 2021-08-23:
Boot to recovery. Mount the esp partition and rename the folder.

salam-siddique, 2021-08-23:
well i could boot into recovery and also mount the esp partition, but i dont know to how to rename the folder.
Could anyone please help?

Thanks