edk2-porting/WOA-Drivers #9: dipper's GPU do not run well.

sssolitude, 2021-08-22:
windows11 ran successfully on my mi8 with the release driver and img this morning ,and about 17:00 I saw the GPU update. So and download the img in Action CI#177 and drivers the latest. I use fastboot boot the newest img and upgrade the drivers using dism++ directly in windows11, then reboot also with the img. but the GPU status is as follows:

Device status:
Windows has stopped this device because it has reported problems. (Code 43)

Device ACPI\VEN_QCOM&DEV_027E&SUBSYS_RENEGA0E&REV_007C\0 had a problem starting.

Driver Name: oem50.inf
Class Guid: {4d36e968-e325-11ce-bfc1-08002be10318}
Service: QCDX
Lower Filters: 
Upper Filters: 
Problem: 0x26
Problem Status: 0xC000038E

TheMorc, 2021-08-22:
the only solution right now that i know of is only reinstalling Windows

TAOcroatia, 2021-08-22:
In devmgmt check the driver version for smmu and iommu. They should be at 1.0.420.0. If not,  install that version and make sure windows doesn't use the newer one