edk2-porting/WOA-Drivers #17: Oneplus 6 showing black screen during installation

leohearts, 2021-09-03:
Time: 2021-09-03/12:00 UTC+8
Newest sources
after dism++ deploy, reboot, it installed like normal, but after an automate reboot during installation it just showed a black screen with LED on.
It may be caused by some changes after https://github.com/edk2-porting/WOA-Drivers/tree/0492fccf7a38d3a3d628e695c5144fb881e9d220 (8 days ago), this version works for me.

leohearts, 2021-09-03:
Yes it works!
I replaced SSDD/wlanmdsp.mbn with the one extracted from stock modem_a image, and installed it.
WiFi now works (but audio and power management doesn't, as before. )

[wlanmdsp.mbn.zip](https://github.com/edk2-porting/WOA-Drivers/files/7106539/wlanmdsp.mbn.zip)

strongtz, 2021-09-21:
Does the problem still occur with latest drivers and UEFI?

leohearts, 2021-09-22:
> Does the problem still occur with latest drivers and UEFI?

Yes.
After some time during installation it turned off
[image]

Jbrimbelibap, 2022-03-06:
I have this same issue, using your wlanmdsp.mbn didn't fix it for me. Should i just try to remove drivers , starting with subsystems/ssdd?

Jbrimbelibap, 2022-03-06:
If it can help anyone, here the wlanmdsp.mbn from my modem_a image
[wlanmdsp.zip](https://github.com/edk2-porting/WOA-Drivers/files/8192923/wlanmdsp.zip)

leohearts, 2022-03-06:
> I have this same issue, using your wlanmdsp.mbn didn't fix it for me. Should i just try to remove drivers , starting with subsystems/ssdd?

Try using the driver from your modem partition?
Sorry I don't exactly know how can I help you…

Jbrimbelibap, 2022-03-06:
Yeah i'm doing this right now, we'll see. If you don't mind uploading your old firware folder with everything working it would be very useful, maybe new versions of the firmware broke more stuff for the chinese version of the phone

Jbrimbelibap, 2022-03-06:
[image]
Using sdm845 generic drivers gets me much further

closed 2022-07-09 by BigfootACA

leohearts, 2023-04-26:
Resolved. It turned out to be the matter of my outdated firmware :(
(I left it with Android 8 for debugging purpose)