edk2-porting/WOA-Drivers #21: Belleryum touchscreen error 12

Tonio1987fr, 2021-09-14:
Hello 
I have error 12 for the touchscreen on my pocophone F1
It's last driver and uefi boot and new windows install
Do you have idea why ?
[image]

Thanks

TheMorc, 2021-09-14:
You also need to download this [devcfg partition](https://cdn.discordapp.com/attachments/736563712466616391/880541358178320454/devcfg.img) and flash it using `fastboot flash devcfg_ab devcfg.img`

Tonio1987fr, 2021-09-16:
I have flash the img file and reinstalled the windows and drivers. I still have the same error.

TheMorc, 2021-09-16:
@Tonio1987fr and do you actually have a FocalTech digitizer on your phone?

Tonio1987fr, 2021-09-17:
How can I see it? i deleted android.
The driver is automatically used, but does not work.

curoviyxru, 2021-09-20:
I confirm that too. My phone has EBBG/FTS panel and I have flashed devcfg image, but it shows error 12.

closed 2021-10-10 by Tonio1987fr