edk2-porting/WOA-Drivers #25: [beryllium]

curoviyxru, 2021-09-20:


curoviyxru, 2021-09-22:
Maybe this will help...
[image]
[image]
[image]
P.S. Windows was installed in Russian, so `User` field is `SYSTEM`.

RusJJ, 2021-09-22:
EFI Image version..?

curoviyxru, 2021-09-22:
> EFI Image version..?

what EFI Image version?

Tonio1987fr, 2021-09-22:
> > EFI Image version..?
> 
> what EFI Image version?

Do you have the latest version?
https://github.com/edk2-porting/edk2-sdm845/releases

I still have the same problem.

curoviyxru, 2021-09-23:
I'm using version 1.0.3 of EFI/drivers.

Tonio1987fr, 2021-09-27:
Have you fixed this problem?

TAOcroatia, 2021-09-27:
> Have you fixed this problem?

Not yet, i'm waiting on a person with fts poco to test some stuff out. Sorry for the wait

TheMorc, 2021-10-06:
> sorry to ask again, but there is any news?

most of us lack the time currently so nothing

TAOcroatia, 2021-11-10:
@Aro2142 @Tonio1987fr @curoviyxru Can you please build the lastest commit of edk2-sdm845 and test if the FTS touchscreen works with it?

curoviyxru, 2021-11-11:
@TAOcroatia I compiled the latest build of edk2-sdm845, reinstalled Windows 10 with latest drivers and it doesn't works.

Luihart, 2022-01-06:
Also tried latest drivers on windows 11. no Touch working

TAOcroatia, 2022-01-06:
I screwed something up trying to support both ebbg and tianma at the same time, gotta fix it

Aro2142, 2022-02-08:
Hello everyone, I have news from a person on the 4pda forum that the touchscreen has worked, but it seems to have an inversion, I think so (he said that the presses do not match).  I personally can’t check, I was unlucky with the display, but there are no errors in the device manager and that’s good.
Tested to work on the latest Uefi and drivers!

Tonio1987fr, 2022-02-08:
> Hello everyone, I have news from a person on the 4pda forum that the touchscreen has worked, but it seems to have an inversion, I think so (he said that the presses do not match). I personally can’t [...]

Thank you for the news. I have since deleted windows from the phone. I might reinstall it one day.

TAOcroatia, 2022-02-08:
> Hello everyone, I have news from a person on the 4pda forum that the touchscreen has worked, but it seems to have an inversion, I think so (he said that the presses do not match).  I personally [...]

So does it all work now? 
As for inversion, try installing the drivers again, it was a mistake in the touch driver. 
Can you link me to the 4pda page?

Aro2142, 2022-02-08:
@TAOcroatia tested by user "istweiter"
 https://4pda.to/forum/index.php?showtopic=884586&view=findpost&p=113075446

RusJJ, 2022-02-08:
You can get a working touch only on FocalTech right now by installing the older UEFI and drivers... There's no working solution for Novatech yet

Aro2142, 2022-02-10:
Asked user istweiter to upload a demo video... As you can see, everything is bad 😹
https://4pda.to/forum/index.php?showtopic=884586&view=findpost&p=113130322

Aro2142, 2022-03-14:
The topic can be closed. The problem with incorrect working of the touch screen is solved by editing the registry.
https://4pda.to/forum/index.php?showtopic=884586&view=findpost&p=113700863 (

ClouDROID-VPN, 2022-03-15:
So the touch issue on poco F1 is totally working on FTS ? how about the process or tutorial to do it 
I want to install win11 on poco f1 but i want to gather more info about the bug and isssues

ClouDROID-VPN, 2022-03-15:
> You can get a working touch only on FocalTech right now by installing the older UEFI and drivers... There's no working solution for Novatech yet

Touch in your phone FTS is working ?

Aro2142, 2022-03-15:
@ClouDROID-VPN Yes, it works perfectly.  But there is a small bug (maybe this is normal): The "double tap" gesture may not work when pressed quickly with a finger, you need to do this with some kind of delay between the first and second tap.  If you have Minecraft Windows 10 edition, you will notice the second bug, I don't know how to explain it.
For other issues, I can’t tell you anything, because this topic is only about the FTS touchscreen.

ClouDROID-VPN, 2022-03-15:
> @ClouDROID-VPN Yes, it works perfectly. But there is a small bug (maybe this is normal): The "double tap" gesture may not work when pressed quickly with a finger, you need to do this with some kind of delay between the first and second tap. If you have Minecraft Windows 10 edition, you will notice the second bug, I don't know how to explain it. `how about the process or tutorial to do it` In my link to the site you will see the attached beryllium_2246x1080_focaltech_fix.reg file, you will only need to open it as an administrator to make corrections to the registry
> 
> For other issues, I can’t tell you anything, because this topic is only about the FTS touchscreen.

Thanks for reply I have FTS but how to flash partition that needs to fix the touch issue

Aro2142, 2022-03-15:
@ClouDROID-VPN The image "devcfg-beryllium_FixTS.img" is written to the partition "devcfg_a" and "devcfg_b" via fastboot.

ClouDROID-VPN, 2022-03-15:
> 

Please give a step by step procedure it will help a lot Thank you very much

ClouDROID-Dev, 2022-03-16:
fastboot flash devcfg_a devcfg-beryllium_FixTS.img
fastboot flash devcfg_b devcfg-beryllium_FixTS.img

or 

fastboot flash devcfg devcfg-beryllium_FixTS.img

Please correct me if im wrong

ClouDROID-Dev, 2022-03-16:
> @ClouDROID-VPN The image "devcfg-beryllium_FixTS.img" is written to the partition "devcfg_a" and "devcfg_b" via fastboot.

Installing FTS on Fastboot mode

What is the correct to flash FTS
fastboot flash devcfg_a devcfg-beryllium_FixTS.img
fastboot flash devcfg_b devcfg-beryllium_FixTS.img

or 

fastboot flash devcfg devcfg-beryllium_FixTS.img

Please correct me if im wrong

Aro2142, 2022-03-16:
This `fastboot flash devcfg_a devcfg-beryllium_FixTS.img
fastboot flash devcfg_b devcfg-beryllium_FixTS.img`

ClouDROID-Dev, 2022-03-16:
> @ClouDROID-Dev This `fastboot flash devcfg_a devcfg-beryllium_FixTS.img` `fastboot flash devcfg_b devcfg-beryllium_FixTS.img`

Can you comment the tutorial/guide how to flash fts to this issue https://github.com/edk2-porting/WOA-Drivers/issues/44 to help the other user that want to flash it Thank you in advance ☺️☺️

Aro2142, 2022-03-16:
@ClouDROID-Dev I don't understand what the problem is😹
 I'll write detailed steps a little later, because now I'm at work

HeroRickyGAMES, 2022-03-29:
> The topic can be closed. The problem with incorrect working of the touch screen is solved by editing the registry. https://4pda.to/forum/index.php?showtopic=884586&view=findpost&p=113700863

can you send me the beryllium_2246x1080_focaltech_fix.reg I can't download at all!

HeroRickyGAMES, 2022-03-29:
from what I'm seeing it's the only thing left to make it really work, I haven't tested it yet, but in the afternoon I'll test it to see if it will work well.

ClouDROID-Dev, 2022-03-29:
> from what I'm seeing it's the only thing left to make it really work, I haven't tested it yet, but in the afternoon I'll test it to see if it will work well.

FTS panel ? if not touch not work

HeroRickyGAMES, 2022-03-29:
> > from what I'm seeing it's the only thing left to make it really work, I haven't tested it yet, but in the afternoon I'll test it to see if it will work well.
> 
> FTS panel ? if not touch not work

That's what they say, I tested flashing the file on the site and it didn't work, I believe that if I use beryllium_2246x1080_focaltech_fix.reg maybe it works

HeroRickyGAMES, 2022-03-29:
> 

A silly question here, how can I know exactly which panel is mine??

HeroRickyGAMES, 2022-03-29:
Dude, I looked at my touch model and bro it's fts_ts theoretically SHOULD work. 
[image]

HeroRickyGAMES, 2022-03-29:
That's good enough

RusJJ, 2022-03-29:
No one cares, can you please stop creating useless notifications?

ClouDROID-Dev, 2022-03-29:
> That's what they say, I tested flashing the file on the site and it didn't work, I believe that if I use beryllium_2246x1080_focaltech_fix.reg maybe it works

Use the latest 1.1.1 drivers and beryllium_2246x1080_focaltech_fix.reg

ClouDROID-Dev, 2022-03-29:
> @ClouDROID-Dev This `fastboot flash devcfg_a devcfg-beryllium_FixTS.img` `fastboot flash devcfg_b devcfg-beryllium_FixTS.img`

follow this instruction

ClouDROID-Dev, 2022-03-29:
> The topic can be closed. The problem with incorrect working of the touch screen is solved by editing the registry. https://4pda.to/forum/index.php?showtopic=884586&view=findpost&p=113700863

The beryllium_2246x1080_focaltech_fix.reg is 404 not found the file is deleted on 4pda forum

Aro2142, 2022-03-29:
You must register on the site 4 pda, after you can download the file

ClouDROID-Dev, 2022-03-29:
> > You must register on the site 4 pda, after you can download this file
> 
> the problem is cant register beccause the captcha is russian cant understand

I talk to my friend russian and help me with captcha now i get the file
https://drive.google.com/file/d/1aqiFd1WMOyR6tGhv54uZs8_EyTzB-fyH/view?usp=sharing

HeroRickyGAMES, 2022-03-29:
> I talk to my friend russian and help me with captcha now i get the file
>
> https://drive.google.com/file/d/1aqiFd1WMOyR6tGhv54uZs8_EyTzB-fyH/view?usp=sharing

Thank you!

HeroRickyGAMES, 2022-03-29:
> > @ClouDROID-Dev This `fastboot flash devcfg_a devcfg-beryllium_FixTS.img` `fastboot flash devcfg_b devcfg-beryllium_FixTS.img`
> 
> follow this instruction

Speaking of expensive installation instructions, the project renegate website should really have a mini tutorial on how to do this...

ClouDROID-Dev, 2022-03-29:
> > > @ClouDROID-Dev This `fastboot flash devcfg_a devcfg-beryllium_FixTS.img` `fastboot flash devcfg_b devcfg-beryllium_FixTS.img`
> > 
> > 
> > follow this instruction
> 
> Speaking of expensive installation instructions, the project renegate website should really have a mini tutorial on how to do this...

Goto closed issue you can see the tutorial for Installing FTS image

HeroRickyGAMES, 2022-03-29:
> Goto closed issue you can see the tutorial for Installing FTS image

Ok

hwanghyunseo, 2022-03-29:
I found the .reg file on usb. Here's a link to see if this helps or not.
https://drive.google.com/file/d/1DdSmiHF4UykejP4W4gCr8xYLY0H9w8uW/view?usp=sharing

HeroRickyGAMES, 2022-03-29:
It's working!! Glad to see it working BUT it looks like it's pretty out of calibrated.
Possibly something in FocalTechTouch.inf, focaltechtouch.cat, FocalTechTouch.sys
I think if we change any lines it should regulate.

ClouDROID-Dev, 2022-03-29:
> It's working!! Glad to see it working BUT it looks like it's pretty out of calibrated. Possibly something in FocalTechTouch.inf, focaltechtouch.cat, FocalTechTouch.sys I think if we change any lines it should regulate.

Can you make a video

HeroRickyGAMES, 2022-03-29:
> > It's working!! Glad to see it working BUT it looks like it's pretty out of calibrated. Possibly something in FocalTechTouch.inf, focaltechtouch.cat, FocalTechTouch.sys I think if we change any lines it should regulate.
> 
> Can you make a video

Sure! Just a second!

Aro2142, 2022-03-29:
Ahh, you need to use beryllium_2246x1080_focaltech_fix.reg

HeroRickyGAMES, 2022-03-29:
> Ahh, you need to use beryllium_2246x1080_focaltech_fix.reg

I used it, but I want to use dnv to see. Need to restart after running beryllium_2246x1080_focaltech_fix.reg?

HeroRickyGAMES, 2022-03-29:
dnv = again.
sorry

HeroRickyGAMES, 2022-03-29:
https://user-images.githubusercontent.com/38869320/160714848-30e42601-e1d5-4213-b0d0-d61cdd9ccc53.mp4

I was listening to music while he was charging a little and I was washing dishes at home, finally working is working well.

Aro2142, 2022-03-29:
`Need to restart after running`
Yes

HeroRickyGAMES, 2022-03-29:
But in the video you can see that it's very unregulated.
I'm going to run beryllium_2246x1080_focaltech_fix.reg again.

HeroRickyGAMES, 2022-03-29:
> `Need to restart after running` Yes

OK thanks!

HeroRickyGAMES, 2022-03-29:
It's working perfectly!!!
Thanks!!
God bless you!!

ClouDROID-Dev, 2022-03-29:
> VID_2022-03-29-10-00-29-656.1.mp4 
> I was listening to music while he was charging a little and I was washing dishes at home, finally working is working well.

The touch is delay

[50 more replies]