edk2-porting/WOA-Drivers #27: Device: chiron (Xiaomi MIX2) is missing

JeromTWinL, 2021-11-09:


tjstyle, 2021-11-09:
Chiron is not supported for now, even in EDK2 stage.

closed 2021-11-09 by tjstyle

JeromTWinL, 2021-11-10:
The device still exists in [edk2-msm8998](https://github.com/edk2-porting/edk2-msm8998) i saw the following device chiron in build.sh, even i built successfully.

JeromTWinL, 2021-11-10:
The following debice chiron is available in edk2, but not in this drivers.

degdag, 2021-11-10:
edk2-msm8998 doesn't have any interesting drivers working yet so be patient please it's a workings times

JeromTWinL, 2021-11-10:
> edk2-msm8998 doesn't have any interesting drivers working yet so be patient please it's a hard workings times making unknown device components working ((no BSP for msm8998 leaked yet))

The following device cheeseburger (OnePlus 5T) with msm8998 soc drivers are leaked,