edk2-porting/WOA-Drivers #29: Poco x3 pro(VAYU)

WtFNickNameLoL, 2022-01-24:
Sorry for the question but i see OnePlus 7 and 7 pro in the Windows Status Matrix(SM8150) that's why i want to ask is there any chance that we will have vayu port in newr feature ?

Aro2142, 2022-01-26:
Hi
As you wrote, the OnePlus 7 has a Snapdragon 8150 (855) processor.  Your phone has a Snapdragon 860 processor, so your device is not expected to be supported anytime soon.  And also there are no hints that this port will be

gitisgithub, 2022-01-26:
> OnePlus 7 has a Snapdragon 8150 (855) processor. Your phone has a Snapdragon 860 processor. 
These CPUs are absolutely similar (only clocks and maybe stepping are different), but both have ACPI table-related problems or something like that. That's why sm8150 repo was removed, I think

WtFNickNameLoL, 2022-01-26:
> Hi As you wrote, the OnePlus 7 has a Snapdragon 8150 (855) processor. Your phone has a Snapdragon 860 processor, so your device is not expected to be supported anytime soon. And also there are no hints that this port will be

 Thanks for the answer, but i just see Xiaomi Pad 5( SM8150AC) in matrix that's why I asking and comment above write about similar structure. Hope driver port will be available anytime soon (in chat they write about  porting problems that's why repo deleted). Good luck and thanks for your work.

strongtz, 2022-01-26:
@WtFNickNameLoL There's no driver for Snapdragon 855 at all. USB is the only thing that will work because Windows has inbox driver for the dwc3 usb controller

WtFNickNameLoL, 2022-01-26:
> @WtFNickNameLoL There's no driver for Snapdragon 855 at all. USB is the only thing that will work because Windows has inbox driver for the dwc3 usb controller

Thanks for the explanation and good luck with your work

closed 2022-01-26 by WtFNickNameLoL