edk2-porting/WOA-Drivers #36: Touch not working (Beryllium). How's it going?

HeroRickyGAMES, 2022-02-23:
How is the touch fix going on Poco F1 (Beryllium)? 

I saw that you need to flash some extra files but they didn't work in my model, I think mine is not the model that uses the EBBG/FTS panel, but just to know, will there be new changes? New drivers? Do you have an idea of this? Just like the folks at Windows Ghost Spectre were pushing to make a new version of Windows Arm customized and optimized, but I still need to study a little more about that.

marohinmark, 2022-02-23:
@HeroRickyGAMES like, does windows ghost spectre even have arm builds

Aro2142, 2022-02-23:
All questions on this topic are already being discussed here: https://github.com/edk2-porting/WOA-Drivers/issues/25 
 FTS touchscreen is now working, but not correctly.  
NVT touchscreen is not supported and is not expected in the near future...

HeroRickyGAMES, 2022-02-24:
I've been thinking and looking at the code, I think if we add a different driver maybe it will work, think with me: if one version works with the driver FingerTipS maybe if we put another version of another device like another xiaomi maybe it will work, I know they use the same processor but if we make something that resembles the own line of the Xiaomi we managed to make it work.

HeroRickyGAMES, 2022-02-24:
And if it doesn't work, we can try to do the same with audio. Audio doesn't work, but if we look for a generic audio driver and test or simply an audio driver from another device we can in theory make it work, I know it's easy to say, but we should test it step by step to see how far it works or not, I want to help with the project! But I promise nothing, because I'm new to programming but I want to learn about it.

closed 2022-03-29 by HeroRickyGAMES