edk2-porting/WOA-Drivers #43: Poco F1 win11 Cellular/Data network is working or not ?

ClouDROID-VPN, 2022-03-15:


ClouDROID-VPN, 2022-03-15:
Who tried to test if the cellular/data network is working ?
 
 Thanks for reply

ShallowAi, 2022-03-15:
According to website said its working but there are two tips you have to read.

If u change sim card slot in windows will croupt EFS/QCN, then LTE not working.

To fix this you need reflash android or restore QCN partition. then reinstall windows and remember not change sim card slot in windows.

hope my poor english can help u.

ClouDROID-VPN, 2022-03-15:
So its working but i cant change the sim card that already inserted ?

ClouDROID-VPN, 2022-03-15:
Someone test it on Windows 10 its totally working fine but i dont know on windows 11 ill wait for someone user of poco f1 win11 before i closed this issue

closed 2022-03-15 by ClouDROID-VPN