edk2-porting/WOA-Drivers #44: Installing FTS devcfg on POCO F1

ClouDROID-Dev, 2022-03-15:
How to install devcfg on POCO F1 without getting error?

Question:
What is the correct process/procedure?
Flashing devcfg before installing windows or flashing devcfg after installing windows ?

Can someone help me with this Thanks in advance

Aro2142, 2022-03-16:
To begin with, two very important questions: 
1) Fastboot unlocked (via Miui flash unlock);  
2) adb & fastboot for Windows did you download?
devcfg is best flashed before installing / booting Windows, otherwise the "blue screen"

ClouDROID-Dev, 2022-03-16:
> 

1. Fastboot unlock - Done
2. Already downloaded

So i flash the devcfg fts first before installing windows ?

Aro2142, 2022-03-16:
> So i flash the devcfg fts first before installing windows ?
Yes

ClouDROID-Dev, 2022-03-16:
> > So i flash the devcfg fts first before installing windows ?
> > Yes

Please send a tutorial please i dont want to face any error thank you

Aro2142, 2022-03-17:
1) Open cmd.exd as administrator;  
2) Write the path to the adb & fastboot directory: "cd C: / adb" (or another path where it is located);  
3) We connect the smartphone in fastboot mode;  
4) In cmd, write two commands: fastboot flash devcfg_a (Image name) AND after fastboot flash devcfg_b (Image name) 
5) If you did not see errors in the transfer, then everything is fine

ClouDROID-Dev, 2022-03-17:
> 1. Open cmd.exd as administrator;
> 2. Write the path to the adb & fastboot directory: "cd C: / adb" (or another path where it is located);
> 3. We connect the smartphone in fastboot mode;
> 4. In cmd, [...]

Thanks for the help i appriciate that.. again Thank you very much.

closed 2022-03-17 by ClouDROID-Dev