edk2-porting/WOA-Drivers #57: MEMORY_MANAGEMENT error

hwanghyunseo, 2022-05-23:
i install on windows 11 poco f1. i used  latest actions file. driver and uefi file. how to fix this error?

enlight3d, 2022-06-02:
reverted to 2.0rc1 and now the BSOD error is "inaccessible boot device"

TAOcroatia, 2022-06-02:
Use [this](https://github.com/edk2-porting/WOA-Drivers/tree/b61155b70e6c67f42fbb7913b6ea97f275c0d118) for now

enlight3d, 2022-06-02:
alright, will test right after. Had to revert to 1.1.1 to stop BSODs... waiting for 1st boot to complete and then I'll try !

kdkdkdkxkzks, 2023-07-11:
> i install on windows 11 poco f1. i used latest file. driver and uefi file. how to fix this error?

[image]