edk2-porting/WOA-Drivers #63: 尊敬的开发者们能适配一下Mix2的驱动吗（msm8998）（谢谢！）

RoyanGG953, 2022-07-09:
费劲力气给一台mix2装上的windows结果没有驱动，真滴难受啊

RoyanGG953, 2022-07-13:
> 实时上是有的，只是没有列出

能在哪里下载或看到吗

qaz6750, 2022-07-14:
[image]
[image]

RoyanGG953, 2022-07-14:
> 实际上因为mix2的usb和pep相互冲突，你只能选择保留其中一个

驱动需要额外修改acpi吗，大概能支持那些驱动？

degdag, 2022-07-14:
Acpi still need a hug changes to be adapted for mi mix2 . at last if you want touchscreen than you need to make the same modifications as been Donne for Mi 6 since they use the same touch ic

1208079702, 2022-08-05:
> > 实际上因为mix2的usb和pep相互冲突，你只能选择保留其中一个
> 
> 驱动需要额外修改acpi吗，大概能支持那些驱动？

有后续嘛？我用的mindows 安装的 请问怎么安装pep驱动呢

closed 2022-08-28 by BigfootACA

zzxxxxxzxxxz, 2025-09-08:
你可以直接套用小米6除了usb之外的所有驱动(包括boot镜像)，pep,触摸屏,蓝牙都可用,GPU无法启用。