edk2-porting/WOA-Drivers #71: Oneplus 6 running very hot (thermal management)

gauravsuman007, 2022-08-28:
My OnePlus 6 runs very hot while using windows 11. I tried enabling battery saver mode, hoping it might reduce cpu utilization and make a difference but it didn't.

Is there any thermal management implemented in the drivers? My hunch is that windows can't read the cpu temperature and hence can't ramp down the cpu power delivery when it's getting hot.

If there's no driver support for thermal management, is there any workaround for the time being to reduce device performance and maybe make it run a little cooler?

Thanks

strongtz, 2022-09-19:
There is thermal management in Windows actually, maybe you want more aggressive policy?

marohinmark, 2022-09-19:
@strongtz It heats the same in android for me

gauravsuman007, 2022-09-19:
> There is thermal management in Windows actually, maybe you want more aggressive policy?

That sounds great. Maybe that will work. I'm a little pessimistic about it as my phone heats up even on very low loads (10% CPU utilisation). This is what led me to believe that windows can't measure the cpu temperature and the throttling happens from chip level ans that means that the device is already too hot by then.

gauravsuman007, 2022-09-19:
> @strongtz It heats the same in android for me

This is different for me. Using Android with web browsing, app installs and such activities keeps the phone cool. Never tried pushing it playing games on Android so can't speak for that. The difference with windows is that the phone heats up in windows even when using file explorer and doing light browsing with edge.