edk2-porting/WOA-Drivers #73: Xiaomi Mi 8 Touchscreen driver stops working

Neo802, 2022-09-02:
Hello there.
I have installed Windows 11 ARM64 on Xiaomi Mi 8 using Project Renegade, flashed the devcfg image into devcfg_a and disabled the power saving features on the Touchscreen driver. It worked fine on most apps and such, however, after a while, it stops working. I have tried multiple times to re-enable touchscreen, either by disabling the touchscreen driver and reenabling it, restarting the phone and so on, however, the touchscreen still remains undetected.

GiaSen, 2022-09-02:
I suggest you to reflash stock devcfg with `fastboot flash devcfg_ab devcfg.img` (you can grab the img from any firmware zip or ROM like my PixelExperience 12.1)
then flash custom one again with same command, yes partition is detected as devcfg_ab

Neo802, 2022-09-02:
Windows apparently is now stuck in a bootloop after flashing the devcfg from Pixel Experience and then the Windows one with the same command.

GiaSen, 2022-09-02:
Oh that happened to me as well when touch wasn't working (from installation) when I flashed in devcfg_a and tried what I said, I had no idea how to fix so I went back to android and reinstall from zero

Neo802, 2022-09-02:
I restarted the phone and now not only does the Windows work fine, but the touchscreen does as well. Thank you.

GiaSen, 2022-09-02:
great

Neo802, 2022-09-02:
Update: Touchscreen again stopped working

strongtz, 2022-09-19:
duplicate with https://github.com/edk2-porting/WOA-Drivers/issues/56

closed 2022-09-19 by strongtz