edk2-porting/WOA-Drivers #76: GPU Driver seems not work on 22H2

lz233, 2022-10-15:


chundk, 2022-10-30:
> device: mix2s

我也遇到了问题，解决方法是手动回滚 Qualcomm System MMU device 驱动到最老的那个版本重启即可

Same issue here, the solution is to roll back the driver of the Qualcomm System MMU device to the old one, take a reboot will make GPU Drivers load properly.

lz233, 2022-10-30:
> > device: mix2s
> 
> 我也遇到了问题，解决方法是手动回滚 Qualcomm System MMU device 驱动到最老的那个版本重启即可
> 
> Same issue here, the solution is to roll back the driver of the Qualcomm System MMU device to the old one, take a reboot will make GPU Drivers load properly.

具体是哪个版本呢

chundk, 2022-10-30:
[image]
看图， 使用老版本的这个