edk2-porting/WOA-Drivers #88: Polaris USB not working

mikotalik, 2022-12-12:
Hello. After 2 full reinstalls, USB on my Xiaomi Mi Mix 2s still doesn't work. I cannot connect mouse, keyboard, or usb flash disk. If I boot with some of the devices attached, I cannot even start Windows. Charging works perfectly.

cocus, 2023-08-13:
What's the old driver pack that "works"? I have installed it recently with the latest edk2-msm (https://github.com/edk2-porting/edk2-msm/releases/tag/2301.1) and latest WOA-Drivers (2210.1-fix), and usb doesn't work (it can charger, but nothing gets detected when I plug a usbc OTG). Funny enough, I see on the device manager that a new device shows up (a usb root hub controller) when I plug the adapter, but nothing gets enumerated. I don't know if something is not turning on the power to the devices or if there's something else going on. I'd like to NOT re-install, is it possible to just use "the old driver release"? If so, can I just downgrade the usb or the drivers related to this issue?

elsa-d55503, 2024-01-08:
> 「動作する」古いドライバー パックは何ですか? [最近、最新の edk2-msm ( https://github.com/edk2-porting/edk2-msm/releases/tag/2301.1](https://github.com/edk2-porting/edk2-msm/releases/tag/2301.1) ) と最新の WOA-Drivers (2210.1-fix) を使用してインストールしましたが、USB がインストールされません。動作します（充電はできますが、USBC OTGを接続しても何も検出されません）。面白いことに、アダプターを接続するとデバイス マネージャーに新しいデバイス (USB ルート ハブ コントローラー) が表示されますが、何も列挙されません。何かが原因でデバイスの電源が入っていないのか、それとも何か他のことが起こっているのかわかりません。再インストールしたくないのですが、「古いドライバー リリース」を使用することは可能ですか? その場合、この問題に関連する USB またはドライバーをダウングレードするだけでよいでしょうか?

> What's the old driver pack that "works"? I have installed it recently with the latest edk2-msm (https://github.com/edk2-porting/edk2-msm/releases/tag/2301.1) and latest WOA-Drivers (2210.1-fix), and [...]

It may be too late now, but USB-OTG worked when I used v2.0rc2 when I installed it.

cocus, 2024-01-24:
> It may be too late now, but USB-OTG worked when I used v2.0rc2 when I installed it.

Yes, I've used the old one. Not sure what changed. But certainly nothing got updated for polaris even after 2210.1-fix :(

jorbig, 2024-05-11:
Maybe this is of help: https://github.com/n00b69/woa-polaris/blob/main/guide/materials.md#toggling-usb-host-mode