edk2-porting/WOA-Drivers #98: Driver Audio for Windows Arm

bangthoyib, 2023-10-14:
How do I create an audio driver from my device for the Windows Arm driver?

SangkarJanda, 2023-10-21:
No response, from everyone.

closed 2023-10-21 by SangkarJanda

qaz6750, 2023-10-21:
Please note that most of the equipment in this project is currently not actively maintained by anyone, and I think that the built-in audio support for SDM845 may never come

SangkarJanda, 2023-10-21:
> Please note that most of the equipment in this project is currently not actively maintained by anyone, and I think that the built-in audio support for SDM845 may never come

No wonder there hasn't been a new update for 1 year, it turns out the Renegade project is dead.

Naletury, 2026-08-07:
i think u could try [AudioRelay](https://audiorelay.net), bluetooth connected to ur headphones or to buy a usb driverless soundbar